radxa-pkg/aic8800 #3: pcie, sdio and usb driver share the same module name `aic8800_fdrv`

amazingfate, 2024-03-10:
I have ported sdio driver to 6.1 kernel: https://github.com/amazingfate/linux-rockchip/commit/0a4d5f9d563f6b80b391fe9bdaad543dc9a64a17.
But I can't install aic8800-usb-dkms after this patch.
```
aic8800_fdrv.ko:
Running module version sanity check.
Module version 6.4.3.0 for aic8800_fdrv.ko
exactly matches what is already found in kernel 6.1.43-vendor-rk35xx.
DKMS will not replace this module.
```

RadxaYuntian, 2024-03-11:
Because there is an aic8800_fdrv.ko for SDIO and aic8800_fdrv.ko for USB. Yes same module name thanks to AIC, which is why it won't be built by dkms. For us though we only install this driver when such AIC device is on-board. Since we don't put 2 variants of this wireless chip on a single product, it is not an issue for us.

FYI they also shares the same firmware path so we had to [patch it](https://github.com/radxa-pkg/aic8800/blob/main/debian/patches/fix-usb-firmware-path.patch) since we only have 1 firmware package.

RadxaYuntian, 2024-03-11:
BTW we have [updated the driver](https://github.com/radxa-pkg/aic8800/commits/main/src) provided by AIC a few times already. For USB I think you really need the latest one as the previous one do not work for us.

RadxaYuntian, 2024-03-11:
I tried to use `DEST_MODULE_NAME[*]` to override the generated module name. However, I see following build errors with the SDIO driver:
```
root@mnt:~# cat  /var/lib/dkms/aic8800-sdio/3.0+git20240116.ec460377-3/build/make.log
DKMS make.log for aic8800-sdio-3.0+git20240116.ec460377-3 for kernel 6.1.43-1-rk2312 (aarch64)
Mon Mar 11 15:54:46 CST 2024
make: Entering directory '/usr/src/linux-headers-6.1.43-1-rk2312'
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: aarch64-linux-gnu-gcc (Debian 10.2.1-6) 10.2.1 20210110
  You are using:           gcc (Debian 12.2.0-14) 12.2.0
make[1]: warning: -j20 forced in makefile: resetting jobserver mode.
  CC [M]  /var/lib/dkms/aic8800-sdio/3.0+git20240116.ec460377-3/build/SDIO/driver_fw/driver/aic8800/aic8800_btlpm/aic_bluetooth_main.o
  CC [M]  /var/lib/dkms/aic8800-sdio/3.0+git20240116.ec460377-3/build/SDIO/driver_fw/driver/aic8800/aic8800_btlpm/rfkill.o
  CC [M]  /var/lib/dkms/aic8800-sdio/3.0+git20240116.ec460377-3/build/SDIO/driver_fw/driver/aic8800/aic8800_fdrv/rwnx_msg_tx.o
  CC [M]  /var/lib/dkms/aic8800-sdio/3.0+git20240116.ec460377-3/build/SDIO/driver_fw/driver/aic8800/aic8800_fdrv/rwnx_msg_rx.o
  CC [M]  /var/lib/dkms/aic8800-sdio/3.0+git20240116.ec460377-3/build/SDIO/driver_fw/driver/aic8800/aic8800_fdrv/rwnx_utils.o
  CC [M]  /var/lib/dkms/aic8800-sdio/3.0+git20240116.ec460377-3/build/SDIO/driver_fw/driver/aic8800/aic8800_fdrv/rwnx_cmds.o
  CC [M]  /var/lib/dkms/aic8800-sdio/3.0+git20240116.ec460377-3/build/SDIO/driver_fw/driver/aic8800/aic8800_fdrv/rwnx_irqs.o
  CC [M]  /var/lib/dkms/aic8800-sdio/3.0+git20240116.ec460377-3/build/SDIO/driver_fw/driver/aic8800/aic8800_fdrv/rwnx_cfgfile.o
  CC [M]  /var/lib/dkms/aic8800-sdio/3.0+git20240116.ec460377-3/build/SDIO/driver_fw/driver/aic8800/aic8800_bsp/aic8800dc_compat.o
  CC [M]  /var/lib/dkms/aic8800-sdio/3.0+git20240116.ec460377-3/build/SDIO/driver_fw/driver/aic8800/aic8800_fdrv/rwnx_strs.o
  CC [M]  /var/lib/dkms/aic8800-sdio/3.0+git20240116.ec460377-3/build/SDIO/driver_fw/driver/aic8800/aic8800_fdrv/rwnx_rx.o
  CC [M]  /var/lib/dkms/aic8800-sdio/3.0+git20240116.ec460377-3/build/SDIO/driver_fw/driver/aic8800/aic8800_bsp/aic8800d80_compat.o
  CC [M]  /var/lib/dkms/aic8800-sdio/3.0+git20240116.ec460377-3/build/SDIO/driver_fw/driver/aic8800/aic8800_bsp/aic_bsp_main.o
  CC [M]  /var/lib/dkms/aic8800-sdio/3.0+git20240116.ec460377-3/build/SDIO/driver_fw/driver/aic8800/aic8800_bsp/aic_bsp_driver.o
  CC [M]  /var/lib/dkms/aic8800-sdio/3.0+git20240116.ec460377-3/build/SDIO/driver_fw/driver/aic8800/aic8800_fdrv/rwnx_txq.o
  CC [M]  /var/lib/dkms/aic8800-sdio/3.0+git20240116.ec460377-3/build/SDIO/driver_fw/driver/aic8800/aic8800_fdrv/rwnx_tx.o
  CC [M]  /var/lib/dkms/aic8800-sdio/3.0+git20240116.ec460377-3/build/SDIO/driver_fw/driver/aic8800/aic8800_bsp/aicsdio.o
  CC [M]  /var/lib/dkms/aic8800-sdio/3.0+git20240116.ec460377-3/build/SDIO/driver_fw/driver/aic8800/aic8800_bsp/aicsdio_txrxif.o
  CC [M]  /var/lib/dkms/aic8800-sdio/3.0+git20240116.ec460377-3/build/SDIO/driver_fw/driver/aic8800/aic8800_fdrv/rwnx_main.o
  CC [M]  /var/lib/dkms/aic8800-sdio/3.0+git20240116.ec460377-3/build/SDIO/driver_fw/driver/aic8800/aic8800_bsp/md5.o
  CC [M]  /var/lib/dkms/aic8800-sdio/3.0+git20240116.ec460377-3/build/SDIO/driver_fw/driver/aic8800/aic8800_fdrv/rwnx_mod_params.o
  CC [M]  /var/lib/dkms/aic8800-sdio/3.0+git20240116.ec460377-3/build/SDIO/driver_fw/driver/aic8800/aic8800_bsp/aicwf_txq_prealloc.o
  CC [M]  /var/lib/dkms/aic8800-sdio/3.0+git20240116.ec460377-3/build/SDIO/driver_fw/driver/aic8800/aic8800_fdrv/rwnx_mesh.o
  CC [M]  /var/lib/dkms/aic8800-sdio/3.0+git20240116.ec460377-3/build/SDIO/driver_fw/driver/aic8800/aic8800_fdrv/rwnx_platform.o
/var/lib/dkms/aic8800-sdio/3.0+git20240116.ec460377-3/build/SDIO/driver_fw/driver/aic8800/aic8800_fdrv/rwnx_mod_params.c: In function ‘rwnx_custregd’:
/var/lib/dkms/aic8800-sdio/3.0+git20240116.ec460377-3/build/SDIO/driver_fw/driver/aic8800/aic8800_fdrv/rwnx_mod_params.c:1731:32: error: ‘REGULATORY_IGNORE_STALE_KICKOFF’ undeclared (first use in this function)
 1731 |     wiphy->regulatory_flags |= REGULATORY_IGNORE_STALE_KICKOFF;
      |                                ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/aic8800-sdio/3.0+git20240116.ec460377-3/build/SDIO/driver_fw/driver/aic8800/aic8800_fdrv/rwnx_mod_params.c:1731:32: note: each undeclared identifier is reported only once for each function it appears in
make[2]: *** [scripts/Makefile.build:250: /var/lib/dkms/aic8800-sdio/3.0+git20240116.ec460377-3/build/SDIO/driver_fw/driver/aic8800/aic8800_fdrv/rwnx_mod_params.o] Error 1
make[2]: *** Waiting for unfinished jobs....
  LD [M]  /var/lib/dkms/aic8800-sdio/3.0+git20240116.ec460377-3/build/SDIO/driver_fw/driver/aic8800/aic8800_btlpm/aic8800_btlpm.o
  LD [M]  /var/lib/dkms/aic8800-sdio/3.0+git20240116.ec460377-3/build/SDIO/driver_fw/driver/aic8800/aic8800_bsp/aic8800_bsp.o
make[1]: *** [scripts/Makefile.build:500: /var/lib/dkms/aic8800-sdio/3.0+git20240116.ec460377-3/build/SDIO/driver_fw/driver/aic8800/aic8800_fdrv] Error 2
make: *** [Makefile:2017: /var/lib/dkms/aic8800-sdio/3.0+git20240116.ec460377-3/build/SDIO/driver_fw/driver/aic8800] Error 2
make: Leaving directory '/usr/src/linux-headers-6.1.43-1-rk2312'
```

Can you show me how did you fix this issue, since your did not have a separate commit for this in your PR.

amazingfate, 2024-03-11:
Just comment that line. Kernel has dropped REGULATORY_IGNORE_STALE_KICKOFF since 6.0.

RadxaYuntian, 2024-03-11:
Added a patch to only enable this for <6.1.0 since we still have older systems to support.

Please check those debs. You should be able to have them all installed on the same system:

[Archive.zip](https://github.com/radxa-pkg/aic8800/files/14555382/Archive.zip)

Firmware structure: https://github.com/radxa-pkg/radxa-firmware/tree/main/firmware/aic8800_fw

amazingfate, 2024-03-11:
Rockchip's 6.1 kernel dropped REGULATORY_IGNORE_STALE_KICKOFF by backporting mainline 6.5: https://github.com/torvalds/linux/commit/e8c2af660ba0790afd14d5cbc2fd05c6dc85e207.
This backport patch is from android: https://github.com/armbian/linux-rockchip/commit/132b7129c5fe90197d2581dd5e786542f9a09d86

RadxaYuntian, 2024-03-11:
Thanks. I'll change the code to detect Rockchip kernel signature later. But for evaluation purpose right now the code should work on your kernel.

commit 75f304cf (2024-03-28): fix: detect Rockchip kernel and their shenanigans

Closes #3.
closed 2024-03-28 by RadxaYuntian