radxa-pkg/aic8800 #8: rock5c onboard wifi: mac address changes after reboot

amazingfate, 2024-04-12:
Tested with v6.1 rk vendor kernel and v6.8.5 mainline kernel.
Mainline kernel need patches to compile, I will create a pr later.

RadxaYuntian, 2024-04-12:
Probably because the eFUSE was not programmed?

amazingfate, 2024-04-12:
It seems like you are maintaining patches in debian/patches, here is my patch to USB driver code to build under kernel v6.8: https://paste.armbian.com/aqazicigaj

amazingfate, 2024-04-12:
> Probably because the eFUSE was not programmed?

There is kernel log: `AICWFDBG(LOGERROR)       no mac address in efuse!`.

RadxaYuntian, 2024-04-12:
> https://paste.armbian.com/aqazicigaj

Rendering `>=` as `≥` is so trippy. That's certainly a bad UX to keep user guessing if someone uses a Unicode keyboard.

RadxaYuntian, 2024-04-12:
[image]
You can just upload the patch to GitHub issue. I does not feel confident to copy past this page into a patch file.

amazingfate, 2024-04-12:
Here is the raw file: https://paste.armbian.com/raw/aqazicigaj

amazingfate, 2024-04-13:
Back to the mac address issue, I tried to set cloned mac address in NetworkManager to get static mac address, but After reboot the mac address is not set:
```
jfliu@rock-5c:~$ ip a show wlx880033771104
3: wlx880033771104: <NO-CARRIER,BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state DORMANT group default qlen 1000
    link/ether 88:00:33:77:11:04 brd ff:ff:ff:ff:ff:ff
jfliu@rock-5c:~$ nmcli -f 802-11-wireless.cloned-mac-address c s 202_5G
802-11-wireless.cloned-mac-address:     88:00:33:77:FC:92
```
I can use this method to let other wifi card like QCA6390 have static mac addredd, but it won't work for aic8800.
Here is the wifi driver log: https://paste.armbian.com/icixiyonir

Anaple, 2024-05-18:
Similarly, I also encountered this problem when using the rk-5.10 kernel.

Is there any solution?

Kernel: https://github.com/Joshua-Riek/linux-rockchip

System version:

Ubuntu22.04

board : rock5 lite

amazingfate, 2024-05-19:
Create an udev rule to get static wifi interface name:
```
$ cat /etc/udev/rules.d/99-radxa-aic8800.rules 
SUBSYSTEM=="net", ACTION=="add", ATTR{address}=="88:00:*", NAME="$ENV{ID_NET_SLOT}"
```
Then use networkmanager to set cloned mac address.

closed 2024-05-19 by amazingfate