radxa-pkg/aic8800 #10: rock-3c （linux 6.6.28）无法安装该deb包

0101wu, 2024-04-24:
系统：armbian（https://www.armbian.com/cf7stdb-entries/rock-3c/） 内核版本：6.6.28
调试任务：AIC8800在armbian系统下不可用
1. 执行命令`sudo dpkg -i aic8800-sdio-dkms_3.0+git20240116.ec460377-6_all.deb ` 安装deb的时候报错：
`/var/lib/dkms/aic8800-sdio/3.0+git20240116.ec460377-6/build/SDIO/driver_fw/driver/aic8800/aic8800_fdrv/rwnx_main.c:6496:22: error: initialization of ‘int (*)(struct wiphy *, struct net_device *, const u8 *, int,  u8,  u8,  u16,  u32,  bool,  const u8 *, size_t)’ {aka ‘int (*)(struct wiphy *, struct net_device *, const unsigned char *, int,  unsigned char,  unsigned char,  short unsigned int,  unsigned int,  _Bool,  const unsigned char *, long unsigned int)’} from incompatible pointer type ‘int (*)(struct wiphy *, struct net_device *, const u8 *, u8,  u8,  u16,  u32,  bool,  const u8 *, size_t)’ {aka ‘int (*)(struct wiphy *, struct net_device *, const unsigned char *, unsigned char,  unsigned char,  short unsigned int,  unsigned int,  _Bool,  const unsigned char *, long unsigned int)’} [-Werror=incompatible-pointer-types]
`
这个错误提示了rwnx_main.c: 6496行在初始化函数指针时发生了不兼容的指针类型错误。
[image]

2. 下载linux6.6.28源码，进入文件include/net/cfg80211.h查看cfg80211_ops的声明，位于4626行：
```
	int	(*tdls_mgmt)(struct wiphy *wiphy, struct net_device *dev,
			     const u8 *peer, int link_id,
			     u8 action_code, u8 dialog_token, u16 status_code,
			     u32 peer_capability, bool initiator,
			     const u8 *buf, size_t len);
```
而驱动源码里，rwnx_cfg80211_tdls_mgmt的函数参数为：
```
static int
rwnx_cfg80211_tdls_mgmt(struct wiphy *wiphy,
	struct net_device *dev,
#if LINUX_VERSION_CODE >= KERNEL_VERSION(3, 16, 0)
	const u8 *peer,
#else
	u8 *peer,
#endif
	u8 action_code,
	u8 dialog_token,
	u16 status_code,
#if LINUX_VERSION_CODE >= KERNEL_VERSION(3, 15, 0)
	u32 peer_capability,
#endif
#if LINUX_VERSION_CODE >= KERNEL_VERSION(3, 17, 0)
	bool initiator,
#endif
	const u8 *buf,
	size_t len)
```
对比两个参数，发现内核cfg80211_ops中tdls_mgmt的参数多了一个 int link_id,故编译报错指针类型不兼容。
3. 我尝试修改驱动源码，将rwnx_cfg80211_tdls_mgmt的函数参数修改为：

```
static int
rwnx_cfg80211_tdls_mgmt(struct wiphy *wiphy,
	struct net_device *dev,
#if LINUX_VERSION_CODE >= KERNEL_VERSION(3, 16, 0)
	const u8 *peer,
#else
	u8 *peer,
#endif
#if (LINUX_VERSION_CODE >= KERNEL_VERSION(6, 6, 28))
	int link_id,
#endif
	u8 action_code,
	u8 dialog_token,
	u16 status_code,
#if LINUX_VERSION_CODE >= KERNEL_VERSION(3, 15, 0)
	u32 peer_capability,
#endif
#if LINUX_VERSION_CODE >= KERNEL_VERSION(3, 17, 0)
	bool initiator,
#endif
	const u8 *buf,
	size_t len)

```
4.修改完源码后，执行命令 sudo dpkg-buildpackage -us -uc -ui ，打算重新制作成deb包，用以安装测试，结果出现报错信息
```
radxa@rock-3c:~/aic8800-3.0-git20240116.ec460377-6$ sudo dpkg-buildpackage -us -uc -ui
dpkg-buildpackage: info: source package aic8800
dpkg-buildpackage: info: source version 3.0+git20240116.ec460377-6
dpkg-buildpackage: info: source distribution jammy
dpkg-buildpackage: info: source changed by "Radxa Computer Co., Ltd" <dev@radxa.com>
dpkg-buildpackage: info: host architecture arm64
 dpkg-source --before-build .
 debian/rules clean
dh clean
   dh_auto_clean
        make -j4 distclean
make[1]: Entering directory '/home/radxa/aic8800-3.0-git20240116.ec460377-6'
rm -rf ./SOURCE
rm -rf debian/.debhelper debian/aic8800-*-dkms debian/aicrf-test debian/debhelper-build-stamp debian/files debian/*.debhelper.log debian/*.*.debhelper debian/*.substvars
make -C src/tools/aicrf_test/ -j4 clean || true
make[2]: Entering directory '/home/radxa/aic8800-3.0-git20240116.ec460377-6'
make[2]: warning: -j4 forced in submake: resetting jobserver mode.
rm *.o wifi_test bt_test
make[2]: Leaving directory '/home/radxa/aic8800-3.0-git20240116.ec460377-6/src/tools/aicrf_test'
make[1]: Leaving directory '/home/radxa/aic8800-3.0-git20240116.ec460377-6'
   dh_clean
 dpkg-source -b .
dpkg-source: error: can't build with source format '3.0 (quilt)': no upstream tarball found at ../aic8800_3.0+git20240116.ec460377.orig.tar.{bz2,gz,lzma,xz}
dpkg-buildpackage: error: dpkg-source -b . subprocess returned exit status 255

```
麻烦 @RadxaYuntian 看看这个报错信息，能否提供下解决思路

RadxaYuntian, 2024-04-24:
我们不是这个驱动的开发厂商啊。驱动不能在主线内核使用这个早就知道了，直接找原厂
另外armbian有使用rockchip内核的版本，那个版本的内核已经集成了aic8800驱动

RadxaYuntian, 2024-04-24:
另外：
1. 看看最新版本的发布日志有提主线内核编译相关的问题：https://github.com/radxa-pkg/aic8800/releases/tag/3.0%2Bgit20240116.ec460377-7
2. 看我们ci是怎么打包的：https://github.com/radxa-pkg/aic8800/blob/main/.github/workflows/release.yml#L24-L27

0101wu, 2024-04-24:
> 另外：
> 
> 1. 看看最新版本的发布日志有提主线内核编译相关的问题：https://github.com/radxa-pkg/aic8800/releases/tag/3.0%2Bgit20240116.ec460377-7
> 2. 看我们ci是怎么打包的：https://github.com/radxa-pkg/aic8800/blob/main/.github/workflows/release.yml#L24-L27

根据这个打包方法成功打包了，现在可以正常驱动wifi了，感谢

RadxaYuntian, 2024-04-24:
实际是从6.5内核开始加入这个参数的：https://github.com/torvalds/linux/commit/c6112046b1a9c130c455ab0bbe74c3f41138693c
如果这么简单就可以用那提个PR吧

0101wu, 2024-05-09:
@RadxaYuntian 您好，我正在制作rock-3c的armbian固件，平时都是在rock-3c上执行命令dpkg安装驱动，现在要把这个aic8800的deb包整合进armbian固件，请问下有什么思路

RadxaYuntian, 2024-05-09:
Armbian应该已经在内核里面集成了aic8800的驱动。关于这个问题你还是最好去armbian社区/discord里面询问

RadxaYuntian, 2024-05-17:
#11 已合并，且包含此修改的新版本已发布。目前先关闭此issue。