radxa-pkg/aic8800 #13: aic8800-usb can't build with kernel 6.9

amazingfate, 2024-06-17:
Upstream commit related: https://github.com/torvalds/linux/commit/b82730bf57b54803ab94abbfd8c4422a7081886d
patch to fix：
```
diff --git a/usr/src/aic8800-usb-3.0+git20240116.ec460377-8/USB/driver_fw/drivers/aic8800/aic8800_fdrv/rwnx_main.c b/usr/src/aic8800-usb-3.0+git20240116.ec460377-8/USB/driver_fw/drivers/aic8800/aic8800_fdrv/rwnx_main.c
index 928b248..cd8ac1c 100644
--- a/usr/src/aic8800-usb-3.0+git20240116.ec460377-8/USB/driver_fw/drivers/aic8800/aic8800_fdrv/rwnx_main.c
+++ b/usr/src/aic8800-usb-3.0+git20240116.ec460377-8/USB/driver_fw/drivers/aic8800/aic8800_fdrv/rwnx_main.c
@@ -1070,9 +1070,7 @@ static void rwnx_csa_finish(struct work_struct *ws)
         } else
             rwnx_txq_vif_stop(vif, RWNX_TXQ_STOP_CHAN, rwnx_hw);
         spin_unlock_bh(&rwnx_hw->cb_lock);
-#if (LINUX_VERSION_CODE >= KERNEL_VERSION(6, 9, 0))
-               cfg80211_ch_switch_notify(vif->ndev, &csa->chandef, 0);
-#elif (LINUX_VERSION_CODE >= HIGH_KERNEL_VERSION3)
+#if (LINUX_VERSION_CODE >= HIGH_KERNEL_VERSION3)
                cfg80211_ch_switch_notify(vif->ndev, &csa->chandef, 0, 0);
 #elif (LINUX_VERSION_CODE >= HIGH_KERNEL_VERSION)
                cfg80211_ch_switch_notify(vif->ndev, &csa->chandef, 0);
@@ -5969,10 +5967,8 @@ int rwnx_cfg80211_channel_switch(struct wiphy *wiphy,
         goto end;
     } else {
         INIT_WORK(&csa->work, rwnx_csa_finish);
-#if (LINUX_VERSION_CODE >= KERNEL_VERSION(6, 9, 0))
-       cfg80211_ch_switch_started_notify(dev, &csa->chandef, 0, params->count, false);
-#elif LINUX_VERSION_CODE >= HIGH_KERNEL_VERSION4
-       cfg80211_ch_switch_started_notify(dev, &csa->chandef, 0, params->count, false, 0);
+#if LINUX_VERSION_CODE >= HIGH_KERNEL_VERSION4
+               cfg80211_ch_switch_started_notify(dev, &csa->chandef, 0, params->count, false, 0);
 #elif LINUX_VERSION_CODE >= HIGH_KERNEL_VERSION2
         cfg80211_ch_switch_started_notify(dev, &csa->chandef, 0, params->count, false);
 #elif LINUX_VERSION_CODE >= KERNEL_VERSION(5, 11, 0)
```

RadxaYuntian, 2024-06-18:
Please retest with the trunk version. The vendor driver has been recently updated.

RadxaYuntian, 2024-06-18:
OK that's why it was tripping. I for sure we did not and do not have code reference 6.9.0.

commit c9baba2e (2024-06-18): fix: update for cfg80211_ch_switch* new parameter list

Closes #13
closed 2024-06-18 by RadxaYuntian