radxa-pkg/aic8800 #19: Did not appear in wifi settings

bongcloudmove, 2024-06-27:
I've installed both drivers from here and manufacturer's website, both seem to be not working, as in the usb is detected in lsusb, but never appear in iwconfig, wifi settings, etc.

here is what I managed to gather

dmesg
```
[44294.824767] usb 1-2: new high-speed USB device number 30 using xhci_hcd
[44294.951321] usb 1-2: New USB device found, idVendor=a69c, idProduct=5721, bcdDevice= 2.00
[44294.951347] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[44294.951356] usb 1-2: Product: Aic MSC
[44294.951363] usb 1-2: Manufacturer: aicsemi
[44294.951370] usb 1-2: SerialNumber: 20200203
[44294.954155] usb-storage 1-2:1.0: USB Mass Storage device detected
[44294.954785] scsi host2: usb-storage 1-2:1.0
[44295.976088] scsi 2:0:0:0: Direct-Access     AIC      flash            1.0  PQ: 0 ANSI: 4
[44295.977042] sd 2:0:0:0: Attached scsi generic sg1 type 0
[44295.977723] sd 2:0:0:0: [sdb] 7648 512-byte logical blocks: (3.92 MB/3.73 MiB)
[44295.977907] sd 2:0:0:0: [sdb] Write Protect is on
[44295.977914] sd 2:0:0:0: [sdb] Mode Sense: 03 00 80 00
[44295.978059] sd 2:0:0:0: [sdb] No Caching mode page found
[44295.978065] sd 2:0:0:0: [sdb] Assuming drive cache: write through
[44295.982187]  sdb: sdb1
[44295.982786] sd 2:0:0:0: [sdb] Attached SCSI removable disk
[44296.059097] usb 1-2: USB disconnect, device number 30
[44296.059414] sd 2:0:0:0: [sdb] tag#0 FAILED Result: hostbyte=DID_ERROR driverbyte=DRIVER_OK cmd_age=0s
[44296.059446] sd 2:0:0:0: [sdb] tag#0 CDB: Start/Stop Unit 1b 00 00 00 02 00
[44296.722736] usb 1-2: new high-speed USB device number 31 using xhci_hcd
[44296.849402] usb 1-2: New USB device found, idVendor=2604, idProduct=0013, bcdDevice= 1.00
[44296.849453] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[44296.849463] usb 1-2: Product: AIC8800DC
[44296.849470] usb 1-2: Manufacturer: AICSemi
[44296.849477] usb 1-2: SerialNumber: 20220127
```

it looks like it's managed to switch properly between mass storage to wifi adapter, but it never manages to function as wifi adapter

on Distributor ID: Ubuntu
Description:    Ubuntu 24.04 LTS
Release:        24.04
Codename:       noble

kernel version 6.8.0-36-generic

content of lsusb
`Bus 001 Device 031: ID 2604:0013 Tenda AIC8800DC`

RadxaYuntian, 2024-06-27:
We do not support Tenda's devices.

RadxaYuntian, 2024-06-27:
Some related Chinese discussion threads:

https://forums.debiancn.org/t/topic/4210/21
https://forums.debiancn.org/t/topic/4630/11

closed 2024-06-27 by RadxaYuntian