radxa-pkg/aic8800 #21: Radxa Zero3W WiFi bus down

csvke, 2024-09-24:
`dmesg` shows
`aicbsp: sdio_err:<aicwf_sdio_bus_pwrctl,1380>: bus down`

`sudo journalctl -u systemd-networkd -f` shows
```
Sep 24 11:45:00 rock-zero3w-ubuntu-rockchip systemd-networkd[734]: wlan0: Configuring with /run/systemd/network/10-netplan-wlan0.network.
Sep 24 11:45:02 rock-zero3w-ubuntu-rockchip systemd-networkd[734]: wlan0: Connected WiFi access point: xxxxxxxxx (xx:xx:xx:xx:xx:xx)
Sep 24 11:45:02 rock-zero3w-ubuntu-rockchip systemd-networkd[734]: wlan0: Gained carrier
Sep 24 11:45:03 rock-zero3w-ubuntu-rockchip systemd-networkd[734]: wlan0: Gained IPv6LL
Sep 24 11:45:05 rock-zero3w-ubuntu-rockchip systemd-networkd[734]: wlan0: DHCPv4 address 192.168.1.140/24, gateway 192.168.1.1 acquired from 192.168.1.1
```

csvke, 2024-09-24:
`dmesg | grep sdio`

```
[   24.899463] aic8800_bsp: loading out-of-tree module taints kernel.
[   24.905075] aicbsp_init
[   24.905107] aicbsp_resv_mem_init 
[   24.972109] aicbsp: aicbsp_set_subsys, subsys: AIC_WIFI, state to: 1
[   24.972114] aicbsp: aicbsp_set_subsys, power state change to 1 dure to AIC_WIFI
[   24.972119] aicbsp: aicbsp_platform_power_on
[   25.032916] aicbsp: aicbsp_sdio_probe:1 vid:0xC8A1  did:0x0082
[   25.038333] aicbsp: aicbsp_sdio_probe:2 vid:0xC8A1  did:0x0182
[   25.038367] aicbsp: aicbsp_sdio_probe after replace:1
[   25.038377] AICWFDBG(LOGINFO)        aicwf_sdio_chipmatch USE AIC8800D80
[   25.038383] aicbsp: aicbsp_get_feature, set FEATURE_SDIO_CLOCK 150 MHz
[   25.038390] aicbsp: aicwf_sdio_reg_init
[   25.053435] AICWFDBG(LOGINFO)        aicbsp: aicbsp_driver_fw_init, chip rev: 7
[   25.053475] rwnx_load_firmware :firmware path = /lib/firmware/aic8800/SDIO/aic8800D80//fw_patch_table_8800d80_u02.bin  
[   25.067822] rwnx_load_firmware :firmware path = /lib/firmware/aic8800/SDIO/aic8800D80//fw_adid_8800d80_u02.bin  
[   25.069846] rwnx_load_firmware :firmware path = /lib/firmware/aic8800/SDIO/aic8800D80//fw_patch_8800d80_u02.bin  
[   25.118734] aicbt_patch_table_load bt btmode[3]:5 
[   25.118765] aicbt_patch_table_load bt uart_baud[3]:1500000 
[   25.118771] aicbt_patch_table_load bt uart_flowctrl[3]:1 
[   25.118777] aicbt_patch_table_load bt lpm_enable[3]:0 
[   25.118782] aicbt_patch_table_load bt tx_pwr[3]:28463 
[   25.125845] aicbsp: bt patch version: - Nov 06 2023 12:44:17 - git 1f5d13b
[   25.126045] rwnx_load_firmware :firmware path = /lib/firmware/aic8800/SDIO/aic8800D80//fmacfw_8800d80_u02.bin  
[   25.243260] AICWFDBG(LOGDEBUG)       aicwf_sdio_probe:1
[   25.243315] AICWFDBG(LOGINFO)        aicwf_sdio_chipmatch USE AIC8800D80
[   25.243322] aicbsp: aicbsp_get_feature, set FEATURE_SDIO_CLOCK 150 MHz
[   25.243329] aicsdio: aicwf_sdio_reg_init
[   25.250389] aicbsp: aicbsp_resv_mem_alloc_skb, alloc resv_mem_txdata succuss, id: 0, size: 98304
[   25.261593] aicbsp: aicbsp_get_feature, set FEATURE_SDIO_CLOCK 150 MHz
[   25.261676] AICWFDBG(LOGTRACE)       >>> rwnx_init_aic()
[   25.262134] AICWFDBG(LOGINFO)        aicwf_prealloc_txq_alloc size is diff will to be kzalloc 
[   25.262170] AICWFDBG(LOGINFO)        aicwf_prealloc_txq_alloc txq kzalloc successful 
[   25.266468] AICWFDBG(LOGINFO)        userconfig file path:aic_userconfig_8800d80.txt 
[   25.266473] AICWFDBG(LOGINFO)        ### Load file aic_userconfig_8800d80.txt
[   25.266481] AICWFDBG(LOGINFO)        rwnx_load_firmware :firmware path = /lib/firmware/aic8800/SDIO/aic8800D80//aic_userconfig_8800d80.txt  
[   25.267796] AICWFDBG(LOGINFO)        ### Load file done: aic_userconfig_8800d80.txt, size=2448
[   25.275927] aicbsp: sdio_err:<aicwf_sdio_bus_pwrctl,1380>: bus down
[ 1427.011479] aicwf_sdio mmc2:390b:1 wlan0: CLOSE
```

RadxaYuntian, 2024-09-25:
Please retest with our system.