radxa-pkg/aic8800 #24: Does not build with kernel 6.12.y+

rpardini, 2025-01-04:
```
   DKMS make.log for aic8800-usb-3.0+git20240327.3561b08f-2 for kernel 6.13.0-rc5-edge-rockchip64 (aarch64)
   Sat Jan  4 14:34:51 UTC 2025
   make: Entering directory '/usr/src/linux-headers-6.13.0-rc5-edge-rockchip64'
   make[1]: Entering directory '/var/lib/dkms/aic8800-usb/3.0+git20240327.3561b08f-2/build/USB/driver_fw/drivers/aic8800'
   make[3]: warning: -j14 forced in makefile: resetting jobserver mode.
   make[4]: warning: -j14 forced in makefile: resetting jobserver mode.
     CC [M]  aic_load_fw/aic_bluetooth_main.o
     CC [M]  aic_load_fw/aicbluetooth.o
     CC [M]  aic8800_fdrv/rwnx_msg_tx.o
     CC [M]  aic_load_fw/aicwf_usb.o
     CC [M]  aic8800_fdrv/rwnx_msg_rx.o
     CC [M]  aic8800_fdrv/rwnx_utils.o
     CC [M]  aic_load_fw/aic_txrxif.o
     CC [M]  aic_load_fw/aicbluetooth_cmds.o
     CC [M]  aic8800_fdrv/rwnx_cmds.o
     CC [M]  aic8800_fdrv/rwnx_irqs.o
     CC [M]  aic_load_fw/aic_compat_8800d80.o
     CC [M]  aic_load_fw/md5.o
     CC [M]  aic_load_fw/aicwf_txq_prealloc.o
     CC [M]  aic8800_fdrv/rwnx_cfgfile.o
     CC [M]  aic8800_fdrv/rwnx_strs.o
     CC [M]  aic8800_fdrv/rwnx_rx.o
     CC [M]  aic8800_fdrv/rwnx_tx.o
     CC [M]  aic8800_fdrv/rwnx_txq.o
     CC [M]  aic8800_fdrv/rwnx_main.o
     CC [M]  aic8800_fdrv/rwnx_mod_params.o
     CC [M]  aic8800_fdrv/rwnx_mesh.o
     CC [M]  aic8800_fdrv/rwnx_platform.o
     CC [M]  aic8800_fdrv/rwnx_pci.o
   aic_load_fw/aicwf_txq_prealloc.c:13:7: warning: no previous prototype for 'aicwf_prealloc_txq_alloc' [-Wmissing-prototypes]
      13 | void *aicwf_prealloc_txq_alloc(size_t size)
         |       ^~~~~~~~~~~~~~~~~~~~~~~~
   aic_load_fw/aicwf_txq_prealloc.c:50:6: warning: no previous prototype for 'aicwf_prealloc_txq_free' [-Wmissing-prototypes]
      50 | void aicwf_prealloc_txq_free(void)
         |      ^~~~~~~~~~~~~~~~~~~~~~~
   aic_load_fw/aicwf_usb.c:1345:5: warning: no previous prototype for 'aicfw_download_fw_8800' [-Wmissing-prototypes]
    1345 | int aicfw_download_fw_8800(struct aic_usb_dev *usb_dev){
         |     ^~~~~~~~~~~~~~~~~~~~~~
   aic_load_fw/aicbluetooth.c:810:6: warning: no previous prototype for 'get_fw_path' [-Wmissing-prototypes]
     810 | void get_fw_path(char* fw_path){
         |      ^~~~~~~~~~~
   aic_load_fw/aicbluetooth.c:818:6: warning: no previous prototype for 'set_testmode' [-Wmissing-prototypes]
     818 | void set_testmode(int val){
         |      ^~~~~~~~~~~~
   aic_load_fw/aicbluetooth.c:822:5: warning: no previous prototype for 'get_testmode' [-Wmissing-prototypes]
     822 | int get_testmode(void){
         |     ^~~~~~~~~~~~
   aic_load_fw/aicbluetooth.c:826:5: warning: no previous prototype for 'get_hardware_info' [-Wmissing-prototypes]
     826 | int get_hardware_info(void){
         |     ^~~~~~~~~~~~~~~~~
   aic_load_fw/aicwf_usb.c:1569:5: warning: no previous prototype for 'aicfw_download_fw' [-Wmissing-prototypes]
    1569 | int aicfw_download_fw(struct aic_usb_dev *usb_dev)
         |     ^~~~~~~~~~~~~~~~~
   aic_load_fw/aicbluetooth.c:831:5: warning: no previous prototype for 'get_adap_test' [-Wmissing-prototypes]
     831 | int get_adap_test(void){
         |     ^~~~~~~~~~~~~
   aic_load_fw/aicbluetooth.c:846:6: warning: no previous prototype for 'get_userconfig_xtal_cap' [-Wmissing-prototypes]
     846 | void get_userconfig_xtal_cap(xtal_cap_conf_t *xtal_cap)
         |      ^~~~~~~~~~~~~~~~~~~~~~~
   aic_load_fw/aicbluetooth.c:859:6: warning: no previous prototype for 'get_userconfig_txpwr_idx' [-Wmissing-prototypes]
     859 | void get_userconfig_txpwr_idx(txpwr_idx_conf_t *txpwr_idx){
         |      ^~~~~~~~~~~~~~~~~~~~~~~~
   aic_load_fw/aicbluetooth.c:886:6: warning: no previous prototype for 'get_userconfig_txpwr_ofst' [-Wmissing-prototypes]
     886 | void get_userconfig_txpwr_ofst(txpwr_ofst_conf_t *txpwr_ofst){
         |      ^~~~~~~~~~~~~~~~~~~~~~~~~
   aic_load_fw/aicbluetooth.c:909:6: warning: no previous prototype for 'rwnx_plat_userconfig_set_value' [-Wmissing-prototypes]
     909 | void rwnx_plat_userconfig_set_value(char *command, char *value){
         |      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
   aic_load_fw/aicbluetooth.c:957:6: warning: no previous prototype for 'rwnx_plat_userconfig_parsing' [-Wmissing-prototypes]
[... 217 lines cut ...]
         | ^~~~~~~~~~~~~~~~
   make[4]: *** [/usr/src/linux-headers-6.13.0-rc5-edge-rockchip64/scripts/Makefile.build:194: aic8800_fdrv/rwnx_main.o] Error 1
   aic8800_fdrv/rwnx_dini.c:105:5: warning: no previous prototype for 'rwnx_cfpga_irq_enable' [-Wmissing-prototypes]
     105 | int rwnx_cfpga_irq_enable(struct rwnx_hw *rwnx_hw)
         |     ^~~~~~~~~~~~~~~~~~~~~
   aic8800_fdrv/rwnx_dini.c:129:5: warning: no previous prototype for 'rwnx_cfpga_irq_disable' [-Wmissing-prototypes]
     129 | int rwnx_cfpga_irq_disable(struct rwnx_hw *rwnx_hw)
         |     ^~~~~~~~~~~~~~~~~~~~~~
   In function 'rwnx_plat_patch_load',
       inlined from 'rwnx_platform_on' at aic8800_fdrv/rwnx_platform.c:2771:11:
   aic8800_fdrv/rwnx_platform.c:1309:9: warning: 'sprintf' argument 3 overlaps destination object 'aic_fw_path' [-Wrestrict]
    1309 |         sprintf(aic_fw_path, "%s/%s", aic_fw_path, "aic8800DC");
         |         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
   aic8800_fdrv/rwnx_platform.c: In function 'rwnx_platform_on':
   aic8800_fdrv/rwnx_platform.c:50:13: note: destination object referenced by 'restrict'-qualified argument 1 was declared here
      50 | extern char aic_fw_path[FW_PATH_MAX_LEN];
         |             ^~~~~~~~~~~
   make[3]: *** [/usr/src/linux-headers-6.13.0-rc5-edge-rockchip64/scripts/Makefile.build:440: aic8800_fdrv] Error 2
   make[2]: *** [/usr/src/linux-headers-6.13.0-rc5-edge-rockchip64/Makefile:1989: .] Error 2
   make[1]: *** [/usr/src/linux-headers-6.13.0-rc5-edge-rockchip64/Makefile:251: __sub-make] Error 2
   make[1]: Leaving directory '/var/lib/dkms/aic8800-usb/3.0+git20240327.3561b08f-2/build/USB/driver_fw/drivers/aic8800'
   make: *** [Makefile:251: __sub-make] Error 2
   make: Leaving directory '/usr/src/linux-headers-6.13.0-rc5-edge-rockchip64'
------------------------------------------------------------------------------------------------------------
```

ModFol, 2025-02-01:
I can confirm that my builds have also failed for both of kernels 6.12 and 6.13

RadxaYuntian, 2025-02-05:
Feel free to submit a PR. Requirement are mentioned in the README.md

closed 2025-02-05 by RadxaYuntian