radxa-pkg/aic8800 #29: [FYI]firmware build failure against LTS linux-6.12.18.

samveen, 2025-03-18:
I'm built and installed the upstream LTS kernel 6.12.18 using https://github.com/radxa-repo/bsp. While installing the built packages, the dkms for the aic8800 (the sdio) version failed as below:

```
<snip>
Setting up linux-image-6.12.18-1-.lts-6.12 (6.12.18-1) ...
dkms: running auto installation service for kernel 6.12.18-1-.lts-6.12:
Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...(bad exit status: 2)
make -j4 KERNELRELEASE=6.12.18-1-.lts-6.12 -C /lib/modules/6.12.18-1-.lts-6.12/build M=/var/lib/dkms/aic8800-sdio/3.0+git20240327.3561b08f-2/build/SDIO/driver_fw/driver/aic8800........................(bad exit s
tatus: 2)
Error! Bad return status for module build on kernel: 6.12.18-1-.lts-6.12 (aarch64)
Consult /var/lib/dkms/aic8800-sdio/3.0+git20240327.3561b08f-2/build/make.log for more information.
</snip>
```

The dkms log is as follows:
```
DKMS make.log for aic8800-sdio-3.0+git20240327.3561b08f-2 for kernel 6.12.18-1-.lts-6.12 (aarch64)
Tue 18 Mar 2025 07:50:21 PM IST
make: Entering directory '/usr/src/linux-headers-6.12.18-1-.lts-6.12'
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: aarch64-linux-gnu-gcc (Debian 10.2.1-6) 10.2.1 20210110
  You are using:           gcc (Debian 10.2.1-6) 10.2.1 20210110
make[2]: warning: -j4 forced in makefile: resetting jobserver mode.
  CC [M]  /var/lib/dkms/aic8800-sdio/3.0+git20240327.3561b08f-2/build/SDIO/driver_fw/driver/aic8800/aic8800_btlpm/aic_bluetooth_main.o
make[3]: warning: -j4 forced in makefile: resetting jobserver mode.
  CC [M]  /var/lib/dkms/aic8800-sdio/3.0+git20240327.3561b08f-2/build/SDIO/driver_fw/driver/aic8800/aic8800_btlpm/rfkill.o
  CC [M]  /var/lib/dkms/aic8800-sdio/3.0+git20240327.3561b08f-2/build/SDIO/driver_fw/driver/aic8800/aic8800_fdrv/rwnx_msg_tx.o
  CC [M]  /var/lib/dkms/aic8800-sdio/3.0+git20240327.3561b08f-2/build/SDIO/driver_fw/driver/aic8800/aic8800_fdrv/rwnx_msg_rx.o
  CC [M]  /var/lib/dkms/aic8800-sdio/3.0+git20240327.3561b08f-2/build/SDIO/driver_fw/driver/aic8800/aic8800_fdrv/rwnx_utils.o
  CC [M]  /var/lib/dkms/aic8800-sdio/3.0+git20240327.3561b08f-2/build/SDIO/driver_fw/driver/aic8800/aic8800_bsp/aic8800dc_compat.o
  CC [M]  /var/lib/dkms/aic8800-sdio/3.0+git20240327.3561b08f-2/build/SDIO/driver_fw/driver/aic8800/aic8800_fdrv/rwnx_cmds.o
/var/lib/dkms/aic8800-sdio/3.0+git20240327.3561b08f-2/build/SDIO/driver_fw/driver/aic8800/aic8800_btlpm/rfkill.c:45:5: warning: no previous prototype for ‘rfkill_bluetooth_init’ [-Wmissing-prototypes]
   45 | int rfkill_bluetooth_init(struct platform_device *pdev)
      |     ^~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/aic8800-sdio/3.0+git20240327.3561b08f-2/build/SDIO/driver_fw/driver/aic8800/aic8800_btlpm/rfkill.c:73:5: warning: no previous prototype for ‘rfkill_bluetooth_remove’ [-Wmissing-prototypes]
   73 | int rfkill_bluetooth_remove(struct platform_device *dev)
      |     ^~~~~~~~~~~~~~~~~~~~~~~
  CC [M]  /var/lib/dkms/aic8800-sdio/3.0+git20240327.3561b08f-2/build/SDIO/driver_fw/driver/aic8800/aic8800_bsp/aic8800d80_compat.o
/var/lib/dkms/aic8800-sdio/3.0+git20240327.3561b08f-2/build/SDIO/driver_fw/driver/aic8800/aic8800_fdrv/rwnx_utils.c:23:5: warning: no previous prototype for ‘rwnx_init_aic’ [-Wmissing-prototypes]
   23 | int rwnx_init_aic(struct rwnx_hw *rwnx_hw)
      |     ^~~~~~~~~~~~~
/var/lib/dkms/aic8800-sdio/3.0+git20240327.3561b08f-2/build/SDIO/driver_fw/driver/aic8800/aic8800_bsp/aic8800dc_compat.c:1697:5: warning: no previous prototype for ‘set_bbpll_config’ [-Wmissing-prototypes]
 1697 | int set_bbpll_config(struct aic_sdio_dev *rwnx_hw){
      |     ^~~~~~~~~~~~~~~~
/var/lib/dkms/aic8800-sdio/3.0+git20240327.3561b08f-2/build/SDIO/driver_fw/driver/aic8800/aic8800_fdrv/rwnx_cmds.c:293:6: warning: no previous prototype for ‘cmd_mgr_task_process’ [-Wmissing-prototypes]
  293 | void cmd_mgr_task_process(struct work_struct *work)
      |      ^~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/aic8800-sdio/3.0+git20240327.3561b08f-2/build/SDIO/driver_fw/driver/aic8800/aic8800_fdrv/rwnx_msg_tx.c:164:18: warning: no previous prototype for ‘rwnx_cmd_malloc’ [-Wmissing-prototypes]
  164 | struct rwnx_cmd *rwnx_cmd_malloc(void){
      |                  ^~~~~~~~~~~~~~~
/var/lib/dkms/aic8800-sdio/3.0+git20240327.3561b08f-2/build/SDIO/driver_fw/driver/aic8800/aic8800_fdrv/rwnx_msg_tx.c:193:6: warning: no previous prototype for ‘rwnx_cmd_free’ [-Wmissing-prototypes]
  193 | void rwnx_cmd_free(struct rwnx_cmd *cmd){
      |      ^~~~~~~~~~~~~
/var/lib/dkms/aic8800-sdio/3.0+git20240327.3561b08f-2/build/SDIO/driver_fw/driver/aic8800/aic8800_fdrv/rwnx_msg_tx.c:203:5: warning: no previous prototype for ‘rwnx_init_cmd_array’ [-Wmissing-prototypes]
  203 | int rwnx_init_cmd_array(void){
      |     ^~~~~~~~~~~~~~~~~~~
/var/lib/dkms/aic8800-sdio/3.0+git20240327.3561b08f-2/build/SDIO/driver_fw/driver/aic8800/aic8800_fdrv/rwnx_msg_tx.c:218:6: warning: no previous prototype for ‘rwnx_free_cmd_array’ [-Wmissing-prototypes]
  218 | void rwnx_free_cmd_array(void){
      |      ^~~~~~~~~~~~~~~~~~~
  LD [M]  /var/lib/dkms/aic8800-sdio/3.0+git20240327.3561b08f-2/build/SDIO/driver_fw/driver/aic8800/aic8800_btlpm/aic8800_btlpm.o
  CC [M]  /var/lib/dkms/aic8800-sdio/3.0+git20240327.3561b08f-2/build/SDIO/driver_fw/driver/aic8800/aic8800_fdrv/rwnx_irqs.o
  CC [M]  /var/lib/dkms/aic8800-sdio/3.0+git20240327.3561b08f-2/build/SDIO/driver_fw/driver/aic8800/aic8800_bsp/aic_bsp_main.o
  CC [M]  /var/lib/dkms/aic8800-sdio/3.0+git20240327.3561b08f-2/build/SDIO/driver_fw/driver/aic8800/aic8800_fdrv/rwnx_cfgfile.o
  CC [M]  /var/lib/dkms/aic8800-sdio/3.0+git20240327.3561b08f-2/build/SDIO/driver_fw/driver/aic8800/aic8800_bsp/aic_bsp_driver.o
  CC [M]  /var/lib/dkms/aic8800-sdio/3.0+git20240327.3561b08f-2/build/SDIO/driver_fw/driver/aic8800/aic8800_fdrv/rwnx_strs.o
  CC [M]  /var/lib/dkms/aic8800-sdio/3.0+git20240327.3561b08f-2/build/SDIO/driver_fw/driver/aic8800/aic8800_fdrv/rwnx_rx.o
  CC [M]  /var/lib/dkms/aic8800-sdio/3.0+git20240327.3561b08f-2/build/SDIO/driver_fw/driver/aic8800/aic8800_bsp/aicsdio.o
[... 198 lines cut ...]
  CC [M]  /var/lib/dkms/aic8800-sdio/3.0+git20240327.3561b08f-2/build/SDIO/driver_fw/driver/aic8800/aic8800_fdrv/rwnx_v7.o
/var/lib/dkms/aic8800-sdio/3.0+git20240327.3561b08f-2/build/SDIO/driver_fw/driver/aic8800/aic8800_fdrv/rwnx_pci.c:85:5: warning: no previous prototype for ‘rwnx_pci_register_drv’ [-Wmissing-prototypes]
   85 | int rwnx_pci_register_drv(void)
      |     ^~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/aic8800-sdio/3.0+git20240327.3561b08f-2/build/SDIO/driver_fw/driver/aic8800/aic8800_fdrv/rwnx_pci.c:90:6: warning: no previous prototype for ‘rwnx_pci_unregister_drv’ [-Wmissing-prototypes]
   90 | void rwnx_pci_unregister_drv(void)
      |      ^~~~~~~~~~~~~~~~~~~~~~~
  CC [M]  /var/lib/dkms/aic8800-sdio/3.0+git20240327.3561b08f-2/build/SDIO/driver_fw/driver/aic8800/aic8800_fdrv/ipc_host.o
/var/lib/dkms/aic8800-sdio/3.0+git20240327.3561b08f-2/build/SDIO/driver_fw/driver/aic8800/aic8800_fdrv/rwnx_dini.c:104:5: warning: no previous prototype for ‘rwnx_cfpga_irq_enable’ [-Wmissing-prototypes]
  104 | int rwnx_cfpga_irq_enable(struct rwnx_hw *rwnx_hw)
      |     ^~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/aic8800-sdio/3.0+git20240327.3561b08f-2/build/SDIO/driver_fw/driver/aic8800/aic8800_fdrv/rwnx_dini.c:128:5: warning: no previous prototype for ‘rwnx_cfpga_irq_disable’ [-Wmissing-prototypes]
  128 | int rwnx_cfpga_irq_disable(struct rwnx_hw *rwnx_hw)
      |     ^~~~~~~~~~~~~~~~~~~~~~
  CC [M]  /var/lib/dkms/aic8800-sdio/3.0+git20240327.3561b08f-2/build/SDIO/driver_fw/driver/aic8800/aic8800_fdrv/rwnx_tdls.o
cc1: some warnings being treated as errors
make[3]: *** [scripts/Makefile.build:229: /var/lib/dkms/aic8800-sdio/3.0+git20240327.3561b08f-2/build/SDIO/driver_fw/driver/aic8800/aic8800_fdrv/rwnx_main.o] Error 1
make[3]: *** Waiting for unfinished jobs....
make[2]: *** [scripts/Makefile.build:478: /var/lib/dkms/aic8800-sdio/3.0+git20240327.3561b08f-2/build/SDIO/driver_fw/driver/aic8800/aic8800_fdrv] Error 2
make[1]: *** [/usr/src/linux-headers-6.12.18-1-.lts-6.12/Makefile:1932: /var/lib/dkms/aic8800-sdio/3.0+git20240327.3561b08f-2/build/SDIO/driver_fw/driver/aic8800] Error 2
make: *** [Makefile:224: __sub-make] Error 2
make: Leaving directory '/usr/src/linux-headers-6.12.18-1-.lts-6.12'
```

I'll try and see what I can do to create a PR with a patch/fix here or in the `bsp` repository, especially if you can assign this issue to me.

RadxaYuntian, 2025-03-19:
Why are you using 3.0+git20240327.3561b08f-2? Try 3.0+git20240327.3561b08f-**4**.

samveen, 2025-03-19:
> Why are you using 3.0+git20240327.3561b08f-2? Try 3.0+git20240327.3561b08f-**4**.

This was the version installed by rsetup -> system-upgrade. The version you are mentinoing is not available in the radxa repositories.

I'll see if I can manually download and install the package.

RadxaYuntian, 2025-03-19:
New package release will go into testing repo first before releasing to production repo: https://radxa-repo.github.io/bullseye-test/files.list

 If you are testing 6.12 kernel you should use testing repo.

samveen, 2025-03-25:
Upgrade to ackage version 3.0+git20240327.3561b08f-4 solves the problem. Closing the issue.

closed 2025-03-25 by samveen