radxa-pkg/aic8800 #31: DKMS error

rahul0x00, 2025-04-14:
Hi, when installing the usb-dkms i'm getting error that dkms is not installed but it is installed

```
(Reading database ... 2194 files and directories currently installed.)
Preparing to unpack aic8800-usb-dkms_3.0+git20240327.3561b08f-4_all(2).deb ...
Unpacking aic8800-usb-dkms (3.0+git20240327.3561b08f-4) over (3.0+git20240327.3561b08f-4) ...
dpkg: dependency problems prevent configuration of aic8800-usb-dkms:
 aic8800-usb-dkms depends on dkms (>= 2.1.0.0); however:
  Package dkms is not installed.

dpkg: error processing package aic8800-usb-dkms (--install):
 dependency problems - leaving unconfigured
Errors were encountered while processing:
 aic8800-usb-dkms
➜   rm aic8800-usb-dkms_3.0+git20240327.3561b08f-4_all\(2\).deb
➜   rm aic8800-usb-dkms_3.0+git20240327.3561b08f-4_all\(1\).deb

```

```
➜   dkms status

aic8800-usb-3.0+git20240327.3561b08f/4: added
vmware-workstation/17.6.3_24583834, 6.6.69-1-lts, x86_64: installed
```

RadxaYuntian, 2025-04-14:
` aic8800-usb-dkms depends on dkms (>= 2.1.0.0); however:`

1. Did you install `dkms` with deb package? Source install does not count.
2. Does your `dkms` deb version >= 2.1.0.0?

rahul0x00, 2025-04-14:
I installed dkms using pacman

RadxaYuntian, 2025-04-14:
Bro...deb is for Debian. To install on Arch you need to repackage it. You can't just install apt/dpkg on Arch and then install package with it.

Check out [`debtap`](https://github.com/helixarch/debtap).

closed 2025-04-14 by RadxaYuntian