radxa-pkg/aic8800 #34: Fails to load for AIC8800DC

joaobal, 2025-05-14:
Hello,

I'm having issues loading the driver for a69c:88dc, on a raspberry pi.

```
pi@raspberrypi:~ $ uname -a
Linux raspberrypi 6.1.21-v7+ #1642 SMP Mon Apr  3 17:20:52 BST 2023 armv7l GNU/Linux
pi@raspberrypi:~ $ lsb_release -a
No LSB modules are available.
Distributor ID:	Raspbian
Description:	Raspbian GNU/Linux 11 (bullseye)
Release:	11
Codename:	bullseye
```

dmesg after mode switch:
```
[   61.269998] usb-storage 1-1.3:1.0: USB Mass Storage device detected
[   61.285010] scsi host0: usb-storage 1-1.3:1.0
[   61.350126] usbcore: registered new interface driver uas
[   62.317749] scsi 0:0:0:0: Direct-Access     AIC      flash            1.0  PQ: 0 ANSI: 4
[   62.320005] sd 0:0:0:0: [sda] 7648 512-byte logical blocks: (3.92 MB/3.73 MiB)
[   62.320580] sd 0:0:0:0: [sda] Write Protect is on
[   62.320610] sd 0:0:0:0: [sda] Mode Sense: 03 00 80 00
[   62.320966] sd 0:0:0:0: [sda] No Caching mode page found
[   62.320990] sd 0:0:0:0: [sda] Assuming drive cache: write through
[   62.325913]  sda: sda1
[   62.327763] sd 0:0:0:0: [sda] Attached SCSI removable disk
[   62.370123] sd 0:0:0:0: Attached scsi generic sg0 type 0
[  111.272667] usb 1-1.3: USB disconnect, device number 4
[  112.076346] usb 1-1.3: new high-speed USB device number 5 using dwc_otg
[  112.207364] usb 1-1.3: New USB device found, idVendor=a69c, idProduct=88dc, bcdDevice= 1.00
[  112.207410] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  112.207432] usb 1-1.3: Product: AIC8800DC
[  112.207450] usb 1-1.3: Manufacturer: AICSemi
[  112.207467] usb 1-1.3: SerialNumber: 20220127
[  432.321729] aic_load_fw: loading out-of-tree module taints kernel.
```

`sudo modprobe aic8800_fdrv_usb`

dmesg after modprobe:
```
[  432.321729] aic_load_fw: loading out-of-tree module taints kernel.
[  432.326434] aic_bluetooth_mod_init 
[  432.326465] RELEASE DATE:2024_0327_3561b08f 
[  432.326673] usbcore: registered new interface driver aic_load_fw

[  432.350796] **********************************************************
[  432.350806] **   NOTICE NOTICE NOTICE NOTICE NOTICE NOTICE NOTICE   **
[  432.350814] **                                                      **
[  432.350821] ** trace_printk() being used. Allocating extra memory.  **
[  432.350828] **                                                      **
[  432.350836] ** This means that this is a DEBUG kernel and it is     **
[  432.350844] ** unsafe for production use.                           **
[  432.350851] **                                                      **
[  432.350858] ** If you see this message and you are not debugging    **
[  432.350866] ** the kernel, report this immediately to your vendor!  **
[  432.350873] **                                                      **
[  432.350880] **   NOTICE NOTICE NOTICE NOTICE NOTICE NOTICE NOTICE   **
[  432.350887] **********************************************************
[  432.353906] AICWFDBG(LOGINFO)	rwnx v6.4.3.0 - 1a4b0054d2M (master)
[  432.353928] AICWFDBG(LOGINFO)	RELEASE DATE:2024_0327_3561b08f 
[  432.354073] AICWFDBG(LOGINFO)	aicwf_usb_chipmatch USE AIC8800DC
[  432.354087] AICWFDBG(LOGINFO)	aicwf_parse_usb endpoints = 4
[  432.354100] AICWFDBG(LOGERROR)	Number of interfaces: 1 not supported
[  432.354110] AICWFDBG(LOGERROR)	AIC8800DC change to AIC8800DW
[  432.354119] AICWFDBG(LOGINFO)	Aic high speed USB device detected
[  432.354128] AICWFDBG(LOGINFO)	aicwf_usb_alloc_rx_urb AICWF_USB_RX_URBS:20 
[  432.354148] AICWFDBG(LOGINFO)	aicwf_usb_alloc_tx_urb AICWF_USB_TX_URBS:200 
[... 131 lines cut ...]
[  437.052252] AICWFDBG(LOGINFO)	start app fail: -110
[  437.052271] AICWFDBG(LOGINFO)	dpd calib fail: -110
[  437.052286] AICWFDBG(LOGERROR)	err_lmac_reqs
[  437.052681] AICWFDBG(LOGINFO)	rwnx_platform_init rwnx_cfg80211_init exit 
[  437.052717] AICWFDBG(LOGERROR)	aicwf_rwnx_usb_platform_init err -1
[  437.052735] AICWFDBG(LOGINFO)	aicwf_bus_deinit Enter
[  437.052750] AICWFDBG(LOGINFO)	aicwf_usb_bus_stop
[  437.052826] AICWFDBG(LOGINFO)	aicwf_usb_rx_complete need to wait for disconnect callback 
[  437.054066] AICWFDBG(LOGINFO)	aicwf_bus_deinit g_rwnx_plat->enabled is false
[  437.054096] AICWFDBG(LOGINFO)	aicwf_bus_deinit stop to bustx_thread!! 
[  437.054196] AICWFDBG(LOGINFO)	usb bustx thread will to stop
[  437.054220] AICWFDBG(LOGINFO)	usb bustx thread stop
[  437.054380] AICWFDBG(LOGINFO)	aicwf_rx_deinit Enter
[  437.054403] AICWFDBG(LOGINFO)	stio rx thread
[  437.054457] AICWFDBG(LOGINFO)	usb busrx thread will to stop
[  437.054477] AICWFDBG(LOGINFO)	usb busrx thread stop
[  437.054571] AICWFDBG(LOGINFO)	usb msg busrx thread will to stop
[  437.054591] AICWFDBG(LOGINFO)	usb msg busrx thread stop
[  437.054730] AICWFDBG(LOGINFO)	aicwf_rx_deinit Exit
[  437.054753] tkn[264]  flags:0032  result: -4  cmd:1037-unknown                  - reqcfm(1038-unknown)
[  437.064261] AICWFDBG(LOGINFO)	aicwf_bus_deinit Exit 
[  437.064729] usb_err:<aicwf_usb_probe,2454>: failed with errno -1
[  437.064789] aic8800_fdrv: probe of 1-1.3:1.0 failed with error -1
[  437.067269] usbcore: registered new interface driver aic8800_fdrv
```

RadxaYuntian, 2025-05-20:
Might be fixed by #35

ZanyRain, 2025-06-01:
I have the same error, and It might be wise to throw this network card in the trash.
There are two point
1. 
````
[  432.354100] AICWFDBG(LOGERROR)	Number of interfaces: 1 not supported
````
https://github.com/radxa-pkg/aic8800/blob/ac5c387ca74e17e91d14d87cd3ba140bb21faf6e/src/USB/driver_fw/drivers/aic8800/aic8800_fdrv/aicwf_usb.c#L2116-L2122

This SKU donot support BT, So I uncomment CONFIG_USB_BT in the Makefile

https://github.com/radxa-pkg/aic8800/blob/ac5c387ca74e17e91d14d87cd3ba140bb21faf6e/src/USB/driver_fw/drivers/aic8800/aic8800_fdrv/aicwf_usb.c#L2564-L2574
https://github.com/radxa-pkg/aic8800/blob/ac5c387ca74e17e91d14d87cd3ba140bb21faf6e/src/USB/driver_fw/drivers/aic8800/aic8800_fdrv/aicwf_usb.h#L20-L31

look at the code logic and modify the aicwf_usb.c to let the driver enabled when CONFIG_USB_BT not define 
2.
````
[  437.037199] cmd_mgr_queue cmd timed-out cmd_mgr->queue_sz:1
[  437.042843] tkn[264]  flags:0032  result: -4  cmd:1037-unknown                  - reqcfm(1038-unknown)
````

This is really confusing, and I have no idea. This card is really cheap.

**If you don't want to throw this network card in the trash**
Aicsemi is a China company, maybe you can try to download [UOS by UnionTech](https://www.chinauos.com/resource/download-professional) or [KylinOS](https://www.cnblogs.com/studyjobs/p/18732929) and install it on the VMs. These two distro is widely used in China government, their official driver must be tested on this two distro. So you can forward the USB card to VM and test if it works.

Driver collection: (include Linux)
https://github.com/peckishrine/aic8800_windows_drivers

joaobal, 2025-06-01:
> I have the same error, and It might be wise to throw this network card in the trash. There are two point 1.

That's what I did. I gave up and bought TP-Link TL-WN823N.

RadxaYuntian, 2025-06-03:
This repository is only for our own (Radxa) products. We are not associated with Aicsemi, and we are not providing support for products manufactured for other companies. We accept community patches to support other use scenarios as a courtesy, but this issue is obviously going nowhere.

Close.

closed 2025-06-03 by RadxaYuntian