radxa-pkg/aic8800 #41: Verify upcoming V4.0_2024_1119_06da8476 release

RadxaYuntian, 2025-07-02:
I just updated the AIC8800 vendor code to a new release, and fixed failed patches along the way. This is currently working on our Cubie A5E running Allwinner 5.15 kernel. My local build is available here:

[aic8800.zip](https://github.com/user-attachments/files/21015180/aic8800.zip)

@CodeChenL please test this with our Rockchip products (SDIO & USB, 5.10 & 6.1).

@amazingfate can you please check with the latest kernel on Armbian to make sure we won't have any regression?

RadxaYuntian, 2025-07-02:
@RadxaStephen is checking if there is a new SDK from AIC, so we might do another update later. I'll let you know once we figure out which version we will use.

CodeChenL, 2025-07-03:
The driver is running normally; I will run iperf3 overnight to test stability

CodeChenL, 2025-07-04:
DKMS error with rock-5c_bookworm_kde_t1
```
rock@rock-5c:~$ cat /var/lib/dkms/aic8800-usb/3.0+git20240327.3561b08f-7/build/make.log
DKMS make.log for aic8800-usb-3.0+git20240327.3561b08f-7 for kernel 6.1.84-2-rk2410 (aarch64)
Fri Jul  4 10:55:30 UTC 2025
make: Entering directory '/usr/src/linux-headers-6.1.84-2-rk2410'
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: aarch64-linux-gnu-gcc (Debian 10.2.1-6) 10.2.1 20210110
  You are using:           gcc (Debian 12.2.0-14) 12.2.0
make[1]: warning: -j8 forced in makefile: resetting jobserver mode.
  CC [M]  /var/lib/dkms/aic8800-usb/3.0+git20240327.3561b08f-7/build/USB/driver_fw/drivers/aic8800/aic_load_fw/aic_bluetooth_main.o
make[2]: warning: -j8 forced in makefile: resetting jobserver mode.
  CC [M]  /var/lib/dkms/aic8800-usb/3.0+git20240327.3561b08f-7/build/USB/driver_fw/drivers/aic8800/aic_load_fw/aicbluetooth.o
  CC [M]  /var/lib/dkms/aic8800-usb/3.0+git20240327.3561b08f-7/build/USB/driver_fw/drivers/aic8800/aic8800_fdrv/rwnx_wakelock.o
  CC [M]  /var/lib/dkms/aic8800-usb/3.0+git20240327.3561b08f-7/build/USB/driver_fw/drivers/aic8800/aic8800_fdrv/rwnx_msg_tx.o
  CC [M]  /var/lib/dkms/aic8800-usb/3.0+git20240327.3561b08f-7/build/USB/driver_fw/drivers/aic8800/aic_load_fw/aicwf_usb.o
  CC [M]  /var/lib/dkms/aic8800-usb/3.0+git20240327.3561b08f-7/build/USB/driver_fw/drivers/aic8800/aic8800_fdrv/rwnx_msg_rx.o
  CC [M]  /var/lib/dkms/aic8800-usb/3.0+git20240327.3561b08f-7/build/USB/driver_fw/drivers/aic8800/aic_load_fw/aic_txrxif.o
  CC [M]  /var/lib/dkms/aic8800-usb/3.0+git20240327.3561b08f-7/build/USB/driver_fw/drivers/aic8800/aic8800_fdrv/rwnx_utils.o
  CC [M]  /var/lib/dkms/aic8800-usb/3.0+git20240327.3561b08f-7/build/USB/driver_fw/drivers/aic8800/aic_load_fw/aicbluetooth_cmds.o
  CC [M]  /var/lib/dkms/aic8800-usb/3.0+git20240327.3561b08f-7/build/USB/driver_fw/drivers/aic8800/aic8800_fdrv/rwnx_cmds.o
  CC [M]  /var/lib/dkms/aic8800-usb/3.0+git20240327.3561b08f-7/build/USB/driver_fw/drivers/aic8800/aic8800_fdrv/rwnx_irqs.o
  CC [M]  /var/lib/dkms/aic8800-usb/3.0+git20240327.3561b08f-7/build/USB/driver_fw/drivers/aic8800/aic_load_fw/aic_compat_8800d80.o
  CC [M]  /var/lib/dkms/aic8800-usb/3.0+git20240327.3561b08f-7/build/USB/driver_fw/drivers/aic8800/aic_load_fw/aic_compat_8800d80x2.o
  CC [M]  /var/lib/dkms/aic8800-usb/3.0+git20240327.3561b08f-7/build/USB/driver_fw/drivers/aic8800/aic8800_fdrv/rwnx_cfgfile.o
  CC [M]  /var/lib/dkms/aic8800-usb/3.0+git20240327.3561b08f-7/build/USB/driver_fw/drivers/aic8800/aic8800_fdrv/rwnx_strs.o
  CC [M]  /var/lib/dkms/aic8800-usb/3.0+git20240327.3561b08f-7/build/USB/driver_fw/drivers/aic8800/aic8800_fdrv/rwnx_rx.o
  CC [M]  /var/lib/dkms/aic8800-usb/3.0+git20240327.3561b08f-7/build/USB/driver_fw/drivers/aic8800/aic_load_fw/md5.o
  CC [M]  /var/lib/dkms/aic8800-usb/3.0+git20240327.3561b08f-7/build/USB/driver_fw/drivers/aic8800/aic_load_fw/aicwf_rx_prealloc.o
  CC [M]  /var/lib/dkms/aic8800-usb/3.0+git20240327.3561b08f-7/build/USB/driver_fw/drivers/aic8800/aic_load_fw/aicwf_txq_prealloc.o
  CC [M]  /var/lib/dkms/aic8800-usb/3.0+git20240327.3561b08f-7/build/USB/driver_fw/drivers/aic8800/aic8800_fdrv/rwnx_tx.o
  CC [M]  /var/lib/dkms/aic8800-usb/3.0+git20240327.3561b08f-7/build/USB/driver_fw/drivers/aic8800/aic8800_fdrv/rwnx_txq.o
  CC [M]  /var/lib/dkms/aic8800-usb/3.0+git20240327.3561b08f-7/build/USB/driver_fw/drivers/aic8800/aic8800_fdrv/rwnx_main.o
  CC [M]  /var/lib/dkms/aic8800-usb/3.0+git20240327.3561b08f-7/build/USB/driver_fw/drivers/aic8800/aic8800_fdrv/rwnx_mod_params.o
  CC [M]  /var/lib/dkms/aic8800-usb/3.0+git20240327.3561b08f-7/build/USB/driver_fw/drivers/aic8800/aic8800_fdrv/rwnx_mesh.o
  CC [M]  /var/lib/dkms/aic8800-usb/3.0+git20240327.3561b08f-7/build/USB/driver_fw/drivers/aic8800/aic8800_fdrv/rwnx_platform.o
  CC [M]  /var/lib/dkms/aic8800-usb/3.0+git20240327.3561b08f-7/build/USB/driver_fw/drivers/aic8800/aic8800_fdrv/rwnx_pci.o
  LD [M]  /var/lib/dkms/aic8800-usb/3.0+git20240327.3561b08f-7/build/USB/driver_fw/drivers/aic8800/aic_load_fw/aic_load_fw.o
/var/lib/dkms/aic8800-usb/3.0+git20240327.3561b08f-7/build/USB/driver_fw/drivers/aic8800/aic8800_fdrv/rwnx_mod_params.c: In function ‘rwnx_set_vht_capa’:
/var/lib/dkms/aic8800-usb/3.0+git20240327.3561b08f-7/build/USB/driver_fw/drivers/aic8800/aic8800_fdrv/rwnx_mod_params.c:936:5: error: this ‘if’ clause does not guard... [-Werror=misleading-indentation]
  936 |     if(rwnx_hw->usbdev->chipid <= PRODUCT_ID_AIC8800D81)
      |     ^~
/var/lib/dkms/aic8800-usb/3.0+git20240327.3561b08f-7/build/USB/driver_fw/drivers/aic8800/aic8800_fdrv/rwnx_mod_params.c:939:9: note: ...this statement, but the latter is misleadingly indented as if it were guarded by the ‘if’
  939 |         band_2GHz->vht_cap.vht_supported = true;
      |         ^~~~~~~~~
cc1: all warnings being treated as errors
make[2]: *** [scripts/Makefile.build:250: /var/lib/dkms/aic8800-usb/3.0+git20240327.3561b08f-7/build/USB/driver_fw/drivers/aic8800/aic8800_fdrv/rwnx_mod_params.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [scripts/Makefile.build:500: /var/lib/dkms/aic8800-usb/3.0+git20240327.3561b08f-7/build/USB/driver_fw/drivers/aic8800/aic8800_fdrv] Error 2
make: *** [Makefile:2015: /var/lib/dkms/aic8800-usb/3.0+git20240327.3561b08f-7/build/USB/driver_fw/drivers/aic8800] Error 2
make: Leaving directory '/usr/src/linux-headers-6.1.84-2-rk2410'
rock@rock-5c:~$ 
```

amazingfate, 2025-07-07:
I can build usb dkms with armbian bookworm 6.1.115.

PR #42 (2025-07-03): fix build for linux 6.16
Fixes #41.
The new driver release has also introduced code need patches for kernel api change in 6.13, 6.14 and 6.15, I have fixed them all.