radxa-pkg/aic8800 #44: tcp_ack_handle_new before ack

oshah81, 2025-07-22:
## Description
With the latest [4.0+git20241119.06da8476-1](https://github.com/radxa-pkg/aic8800/releases/tag/4.0%2Bgit20241119.06da8476-1) aic8800 driver, the below message is filling up my dmesg logs.

```
[705.582490] tcp_ack_handle_new before ack: 1955003393, 1954994825
[705.582529] tcp_ack_handle_new before ack: 1955003393, 1954996253
[3645.782766] tcp_ack_handle_new before ack: 1903582389, 1903570709
[3645.782802] tcp_ack_handle_new before ack: 1903582389, 1903575089
[3645.782812] tcp_ack_handle_new before ack: 1903582389, 1903576549
[3650.331694] tcp_ack_handle_new before ack: 1121542648, 1121541248
[3651.378001] tcp_ack_handle_new before ack: -1618919653, -1618919654
[3752.938276] tcp_ack_handle_new before ack: -1953319908, -1953331219
[3752.938316] tcp_ack_handle_new before ack: -1953319908, -1953329604
[3752.938321] tcp_ack_handle_new before ack: -1953319908, -1953321368
...
```

The above messages seem to occur when opening new tcp connections (not 100% of the time, but often enough for it to cause a degraded wifi experience). It more frequently occurs during heavy network usage (when opening tabs, or mass reloading pages), where the above message can occur once every second.

As a result, whenever the above message pops up in dmesg, I'm seeing wifi connectivity problems (long delays, retried connections, "this site cannot be reached" errors).

## Hardware
* Device: Radxa Rock 5C Lite
* Wifi chip: Built-in AIC8800D80 USB
* OS: Armbian v25.5.1 for Rock 5C running Armbian Linux 6.16.0-rc3-edge-rockchip64
* Armbian Edition: minimal Ubuntu
* Wifi Connection: Wifi 5 connection to my router (Virgin Media superhub 3)
* Build: 4.0+git20241119.06da8476-1

**iwconfig** result

```
wlan0     IEEE 802.11ac  ESSID:"****"  Nickname:"AIC@8800"
Mode:Auto  Frequency:5.22 GHz  Access Point: **:**:**:**:**:**
Bit Rate:195 Mb/s
Encryption key:off
Link Quality=0/100  Signal level=-57 dBm  Noise level=0 dBm
Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
Tx excessive retries:45  Invalid misc:0   Missed beacon:0
```

## Possible Cause
The message has been traced to `ln 486` of `src/USB/driver_fw/drivers/aic8800/aic8800_fdrv/aic_tcp_ack.c` -> `tcp_ack_handle_new()`

## Steps to Reproduce
1. Have a device (like the Rock 5C) that runs the AIC8800 hardware
2. Install the 4.0+git20241119.06da8476-1 driver. I also tried building from the latest repository - same result.
3. Connect the AIC8800 to the wifi router (I use `nmtui`).
4. Create some network requests on the device (in my case, I did some web browsing).
5. Check `dmesg` for output

## More information
`iwconfig` shows a nonzero value for tx excessive retries.

This error did not occur with [3.0+git20240327.3561b08f-7](https://github.com/radxa-pkg/aic8800/releases/tag/3.0%2Bgit20240327.3561b08f-7) (however, due to other instability reasons related to resetting the wifi radio, I cannot rollback to the previous version).

oshah81, 2025-07-24:
Thanks for updating the code. The latest sdk code did _not_ fix the bug.

What did fix it though, was disabling the `CONFIG_FILTER_TCP_ACK` feature in `‎src/USB/driver_fw/drivers/aic8800/aic8800_fdrv/Makefile`.

## Change

1. Open up `‎src/USB/driver_fw/drivers/aic8800/aic8800_fdrv/Makefile`
2. Go to Line 153 which currently reads
```
CONFIG_FILTER_TCP_ACK =y
```
3. Change the line to read
```
CONFIG_FILTER_TCP_ACK =n
```
4. Rebuild, reinstall and reboot.

And the errors are gone and my wifi connection is stable.
This flag also exists with the SDIO and PCIE drivers, but since I don't have those hardware, I cannot test to confirm if the flag should also be disabled for those hardware.

RadxaYuntian, 2025-07-25:
Thanks for the finding. If we can confirm this on our USB devices then we will look to make a new release with this fix included.